WYVERN2742/Peanut #3: Initial Component Selection

WYVERN2742, 2024-05-05:


WYVERN2742, 2024-05-05:
@Brandagot To do sourcing on side-mounted switches
@WYVERN2742 To do sourcing on footprints, and compile datasheets

WYVERN2742, 2024-05-06:
Managed to get a working easyeda -> kicad footprint workflow, main limitations are models for strange parts, and some footprints

WYVERN2742, 2024-05-07:
Got datasheets, footprints, and models for all ✔'s in the component table above.
Just waiting for buzzer, switch and ADC sourcing

Turns out, the pressure sensor works via a [Wheatstone Bridge circuit](https://meritsensor.com/the-wheatstone-bridge/), so it gives `Vout-` and `Vout+`. The differences are too small for direct ADC, so we need to boost them with an opamp.

WYVERN2742, 2024-05-12:
Added 12-bit ADC [ADS1015IDGSR](https://www.lcsc.com/product-detail/Analog-To-Digital-Converters-ADCs_Texas-Instruments-ADS1015IDGSR_C193969.html) and IntAmp [INA350ABSIDDFR](https://www.lcsc.com/product-detail/Instrumentation-OpAmps_Texas-Instruments-INA350ABSIDDFR_C5184576.html) for pressure sensing.

Need to collate I2C addresses, and figure out how to wire battery reading;- we may need another i2c muxer for more outs...

Brandagot, 2024-06-09:
[MCP4728T-E/UN](https://www.lcsc.com/product-detail/Digital-To-Analog-Converters-DACs_Microchip-Tech-MCP4728T-E-UN_C478093.html), play with LDAC / UDAC settings when we get a dev board. Pulling LDAC low will mean immediate output when sending data.

WYVERN2742, 2024-06-09:
~£142.61 without vat
~£171.13 with vat ex delivery

WYVERN2742, 2024-06-09:
Current cost is roughly ~$28.94 / board (or £22.86) without PCB or assembly.

Brandagot, 2024-06-09:
Need a through-hole right angle push button, instead of surface mount.

WYVERN2742, 2024-06-09:
Will bulk-order and manually solder through-hole components, to save on assembly cost

WYVERN2742, 2024-06-17:
Changed from [PCF8574ADWR](https://www.lcsc.com/product-detail/I-O-Expanders_Texas-Instruments-PCF8574ADWR_C2867583.html) -> [PCF8574APWR](https://www.lcsc.com/product-detail/I-O-Expanders_Texas-Instruments-PCF8574APWR_C94361.html) (Package change, it's now smaller, in stock and cheaper!)

WYVERN2742, 2024-06-17:
Just noticed: We forgot about J1 & J2 Connectors for pinenut and extension boards :/